profi200/open_agb_firm #115: Fatfs disk not ready

ZekeN618, 2023-06-03:
I just installed the latest version of open agb to my red and white NSMB2 Nintendo 2DS and when I attempt to boot the firmware using Luma3DS it just launches to a black screen that says "Error: fatfs fisk not ready"
Write protection is not enabled on my SD card

profi200, 2023-06-06:
With latest you mean from the Actions tab above?

ZekeN618, 2023-06-06:
No, I used the one from here: https://github.com/profi200/open_agb_firm/releases/latest which I assumed was best to use since its the link on the main page?

profi200, 2023-06-06:
Try the latest nightly from the Actions tab.

ZekeN618, 2023-06-06:
> Try the latest nightly from the Actions tab.

If you're referring to the one here: https://github.com/profi200/open_agb_firm/actions/runs/4856756922, I have now and it still gives me the fatfs disk not ready error