profi200/open_agb_firm #118: is this still maintained?

GreatNateDev, 2023-06-24:


profi200, 2023-06-30:
Sort of. I didn't have much motivation to work on it in the recent months.

GreatNateDev, 2023-08-31:
:( well if you ever do it again can't wait for cheat support

closed 2024-08-12 by GreatNateDev

GreatNateDev, 2024-11-03:
seems as though this is very active and definitely maintained loved that the dev got his motivation back ❤️

GreatNateDev, 2025-08-24:
Bump! I'm still using this is this deprecated for another form or is this what everyone is still using?

GreatNateDev, 2025-08-29:
> There is no alternative as far as i'm aware. Currently there is more important stuff i invest time in which is why not much is happening.

ok im glad you still acknowledged it. because a-lot of people will just leave a project to die without any communication.