profi200/open_agb_firm #123: Boktai doesn't save time and date

pitrako, 2023-07-24:
Any way of making RTC work on this game?

profi200, 2023-07-24:
Currently no. This is explained in the README. The hardware only supports the RTC in combination with flash save types but this game uses EEPROM.

Rantunah, 2024-07-21:
I've been trying to make this game work with RTC correctly by patching it to save to FLASH512 and FLASH1M, manually editing the header, and setting the per game `saveType` setting to either of the RTC options of flash.
The game boots and prompts me to set up the time with the correct date and time already pre-filled. After that it fails to save.
I've tried using these tools to patch the game. All with the same result stated above:
- [gba-flash-patcher](https://github.com/metroid-maniac/gba-flash-patcher/tree/main)
- [Flash1M_Repro_SRAM_Patcher](https://github.com/bbsan2k/Flash1M_Repro_SRAM_Patcher)
- [gba-auto-batteryless-patcher](https://github.com/metroid-maniac/gba-auto-batteryless-patcher)

I realize that this might not be an agb_firm issue but, do you have any idea if it is at all possible to patch Boktai to save the same way as Pokémon games for instance? These work fine. Even the Pokémon Unbound Romhack has a correct time reading.

profi200, 2024-07-21:
It's definitely possible but i don't know if these tools work for Boktai. The tools for bootleg patching definitely will not because they save on the "ROM" (a flash chip) instead of external flash.

Rantunah, 2024-07-21:
Ok, so I thought these tools just changed the save type and that I'd need to match open_agb_firm configuration for the ROM to work.
Can you briefly describe to me what one would need to do or look into to patch the game to work correctly with agb_firm?
I'm down for some hex copy/pasting between Boktai and a working game it that's all there is to it.

profi200, 2024-07-21:
All EEPROM related functions must be replaced by equivalent flash functions and these functions must execute from internal RAM. That's about it.

That gba-flash-patcher seems to search for areas with many zero bytes in the ROM. I'm not sure how this is supposed to work if it's running from ROM because gbatek mentions that you must not use any flash read/write/erase functions from ROM.