profi200/open_agb_firm #130: The X Button's Unique Feature...

NateTwoo, 2023-08-23:
Any way to disable to LCD backlight toggle the X button is associated with? I want to use X as a B button and the screen just blacking out when I just want to move is very inconvenient...

profi200, 2023-08-23:
To be honest i don't know how to resolve the situation for now. If i remove the brightness adjustment feature others will complain at me. I might add more button mappings. I dunno.

Arkham-Slayer, 2023-08-25:
If I may add, given there are technically two hotkeys, Y currently only doing screen-dumps and X doing all the other screen functions, I think it could work if you could merge them into Y for now to free up X for this situation, unless you have other plans for Y. It could also make remapping the hotkey function easier since you only have one hotkey to work with if you ever want to implement that as a feature.

profi200, 2023-08-26:
The real plan is making them a menu option but until then i have to work around these limitations.

NintendoManiac64, 2023-10-13:
Is there no way to make it use the Start button instead of X and Select instead of Y?  This would then match the mapping for changing brightness when in DS mode (start + up/down).

Or if you want something even safer, Select instead of X (and then subsequently Start instead of Y) since Select is on the left side already as it also was on the Game Boy, therefore I'm uncertain if there are any games at all that require the user to hold Select and a D-pad direction at the same time.

　

> Y for B is more popular

Is it though? I can't help but wonder if it's simply those people being very vocal about it or people that specifically grew up with an NES and/or SNES since, as someone that even grew up with a Game Boy rather than an NES or SNES, I always found B + Y to be maddening and found A + B to make perfect sense.as it most similarly matches the Game Boy's button placement and generally relies on the same muscle memory (I ran into the opposite issue with Contra 4 where it used B + Y and was therefore almost kind of unplayable for me due to tons of A + B muscle memory over the years)