profi200/open_agb_firm #131: .sav doesn’t work with pksm a pkhex port for 3ds gonna try doing it in basic pkhex if not then I will try and get in contact with pksm dev

btl200, 2023-08-25:


btl200, 2023-08-25:
To add on to this I tried booting pokemon saphire on a different emulator and it didn’t want to use the save from this emulator

btl200, 2023-08-25:
Still haven’t tried useing pkhex on pc yet but I just got home

Dartz150, 2023-08-25:
From the readme page:

```
Most emulators output EEPROM saves differently than what open_agb_firm expects, making them incompatible. Fortunately, they are very easy to fix, using https://exelotl.github.io/gba-eeprom-save-fix/ by exelotl.

The tool also works vise versa, if you want to use a save generated by open_agb_firm with an emulator.
```

So you should convert your save back and forth to load it into pkhex to fix this issue.

btl200, 2023-08-26:
Ok good to know thanks

btl200, 2023-08-26:
Question could any of the advanced options make it so I won’t have to change the .save type in the future and if so could you help me walk through doing it

btl200, 2023-08-26:
Never mind I have 100 percent figured it out

profi200, 2023-08-26:
For the record:
Pokemon games use flash savegames. They need no conversion. And the savegames are located in "/3ds/open_agb_firm/saves".

closed 2023-08-26 by profi200