profi200/open_agb_firm #132: Stuck on yellow text menu

ChefSinkz, 2023-08-29:
I installed the FIRM file and now every time I boot up the system it'll just send me to the file menu and I can't get to the actual home menu. Did I do something wrong in the process?

profi200, 2023-08-29:
What yellow text menu? Assuming oaf always boots it sounds like one of your buttons is stuck.

ChefSinkz, 2023-08-29:
it essentially automatically boots up to this[image]

profi200, 2023-08-30:
Either you replaced the boot.firm file on the root of your SD card (replace it with Luma3DS again) or your START button is stuck and always triggers Lumas bootloader.

ChefSinkz, 2023-08-30:
I replaced Luma3DS with the newest version and still nothing. I can't really do much with these menus, I'll open the folder where I put the new boot.firm file and nothing.