profi200/open_agb_firm #134: Causes Stuck Pixels

CarlBagel, 2023-09-17:
I installed open_agb_firm today. I was testing it with "Classic NES Series - Super Mario Bros.". I left the DS closed for about 15 minutes. When I came back, parts of the brick floor and the outline of a cloud refused to update. I restarted the DS, and the pixels were still stuck. (See picture – white horizontal dashes across the bottom of the screen.) I shut off the DS, waited 5 minutes, and rebooted. The pixels were unstuck.

[image]

profi200, 2023-09-17:
IPS LCDs are notorious for this temporary "burn in" effect. It will go away after a hour or so. It's caused by 60 Hz flickering graphics like a lot of GB and GBA games use. There is nothing i can do at the moment but inter frame blending is planned which will fix this at the cost of slightly more motion blur.

Maesejesus, 2023-10-13:
I know this is a month old, but from memory, the flickering is very common with the GBA NES Classics, because they squish what were 224 vertical pixel "tall" games down to the 160 vertical pixels of the GBA, and instead of doing something a bit more... of an effort, let's say it that way, they just squished the picture, so the different vertical lines try to fight between eachother to be viewable.

Very few games of that line did not suffer from that issue, and Super Mario Bros is one is one of the worst examples of that.

This shouldn't be an issue playing normal GB/GBA games as flickering elements last very tiny amounts of time and tend to use very little screen space, not to mention they are usually moving sprites (like projectiles), but with these ports thanks to what I said, the flickering graphics stay all the time on the same points of the screen.

In all honesty, it's not an issue of OPEN_AGB_FIRM at all, but those shoddy ports. It would happen if playing them by other means.

profi200, 2023-10-13:
I don't think it's shoddy ports. This is simply how the hardware works. There is no transparency so game devs had to get creative. They choose to abuse the slow pixel response time of the LCD as a result. This trick has been used since the very first Game Boy.

Maesejesus, 2023-10-14:
I am sorry but I disagree, when you compare between some of these ports with the rest, they seem much lower quality.

Compare Super Mario Bros with, I don't know, Xevious, Dr. Mario or Castlevania. There's no flickering. And while yes, the viewable area is smaller than on NES, they look fine and they're playable, don't harm the screen nor your eyes. 

I will admit though that after trying more of these again, there's less of the bad ports with these flickering graphics than I remember, but those are still there, like Excitebike or Balloon Fight from the ones I've checked.

In fact, they must have known how bad the flickering was because for the release of Super Mario Bros. 2 they fixed it. Maybe it was the first few they did that have this particular thing going on.

There have been games to use flickering as a trick in many games before, it's a classic staple when trying to improve performance or create some visual effect, but these specific examples don't feel at all like any of that, just carelessness to me.

But regardless of any of this, my intention was to simply say that this wasn't OPEN_AGB_FIRM's fault.