profi200/open_agb_firm #137: Mother 3 deletes save file

epicwappy20, 2023-10-07:
The fan translation of mother 3 was unable to save properly, and the .sav is for a chapter 2 save, even though i made it to chapter 4

profi200, 2023-10-07:
Use a clean, verified ROM together with the patch file. A prepatched ROM will cause a fallback to inaccurate save type detection.

epicwappy20, 2023-10-07:
Do I put the patch file in the same directory as the ROM, or am I just supposed to patch the rom.

profi200, 2023-10-07:
Same directory as the ROM and same name (file extension differs).

epicwappy20, 2023-10-07:
> Same directory as the ROM and same name (file extension differs).

I put the UPS in the same directory and it loaded the unpatched version

epicwappy20, 2023-10-07:
The patch is only available in UPS format

epicwappy20, 2023-10-07:
the files are: Mother3.gba(verified checksum) Mother3.sav, and Mother3.ups

epicwappy20, 2023-10-07:
when booting the game in MGBA the patch loads

epicwappy20, 2023-10-07:
Looks like I just needed to update open_agb_firm

closed 2023-10-07 by epicwappy20