profi200/open_agb_firm #139: Soft-patching of BPS patches?

NintendoManiac64, 2023-10-14:
Soft-patching of BPS patches exists in various emulators that typically include the very same IPS and UPS that open_agb_firm already supports.

- [RetroArch] https://docs.libretro.com/guides/softpatching/

> currently supports UPS, IPS and BPS patching formats

　
- [Snes9x GX] https://github.com/dborth/snes9xgx/releases/tag/4.3.8

> Add BPS soft-patching support (thanks qwertymodo!)

profi200, 2023-10-14:
The problem with BPS is that it's a huge pain to implement and if i recall correctly there are even unofficial formats to combine multiple BPS patches into one file.

I don't know when i will be able to support it.

dudamoos, 2026-10-15:
Any news on this? If I recall correctly it's a fairly simple format (documented [here](https://floating.muncher.se/byuu/bps/bps_spec.html)). It's become the preferred patch format for many ROM hacks. 

The "multiple files" version was a separate format (with a separate file extension - BPM) that the original creator abandoned. As far as I know no one uses it.