profi200/open_agb_firm #140: [Feature Request] Add setting to disable RTC

Musilitar, 2023-10-20:
Hi,

First of all thanks so much for this awesome piece of software!
The latest update with the button remapping is amazing!

I have a pretty niche feature request, but if it's not too much work it would be great to have.
The idea is to have a setting that disables the RTC, to simulate games' batteries running dry.
Ideally this would be a per-game setting, but a global one would work as well.
The use case is that in Pokemon Ruby & Sapphire you can do RNG manipulation, but this only works if the game detects a dry battery.

Thanks for considering and all the continued hard work!

papajefe, 2024-03-26:
Second this one. Would be very useful for Ruby\Sapphire RNG manipulation to be able to toggle RTC battery on\off with a per-game setting.

profi200, 2024-03-27:
You can force the save type to a non-RTC one via a per-game config file if i recall correctly. Alternatively you can edit the save type db but this will make all copies of the same game behave the same.