profi200/open_agb_firm #143: Ms. Pac-Man - Maze Madness turns its own in-game audio settings to off on each boot

Dartz150, 2023-10-23:
**Game**
Ms. Pac-Man - Maze Madness | BPCE, BPCP | USA, EUR

**Device Info**
Commit: 63e4422

Console: New3DS

Issue
The game disables audio in its own settings every time you load the game, even with previously saved data with sound enabled.

Screenshot taken from no$GBA just for reference, the issue doesn't happen on it.

[image]

Dartz150, 2023-10-23:
Turns out that I can only reproduce this in one specific 3DS console (my KTR model), the game works as intended in my other two 3DS consoles, an o3DS and a new3DSXL.

 So it could be an FCRAM issue. I'll reflow the chip later and report the result back, for the sake of FCRAM issues documentation.

profi200, 2023-10-23:
Just to add what i already said in chat. I can't reproduce it with and without BIOS intro. Latest commit with the EUR version and fresh save file.

Dartz150, 2024-01-09:
Turns out that this issue was caused by a faulty FCRAM, which also made my KTR n3DS to fail with other software in 3DS mode, leading to random crashes. I had to make a reflow job to the FCRAM chip and this fixed any issue completely, both on oaf and 3DS mode, which means that my FCRAM chip may had cracked solder balls.

Some pictures from the process:

[image]

[image]

[image]

After cleaning the chip:

[image]

The console fully working in both oaf (Ms. Pacman has sound again)
[image]

[image]

closed 2024-01-09 by Dartz150