profi200/open_agb_firm #145: Savegame differences across regions?

CobaltoSketcherino, 2023-11-03:
The USA versions of _Klonoa: Empire of Dreams_ and _Klonoa 2: Dream Champ Tournament_ aren't saving correctly. The odd part is that the European release of _Empire_, and the Japanese release of _DCT_ save perfectly fine. Is there an inconsistency with save types across regions?
What should I do to help resolve this?

profi200, 2023-11-03:
Make sure the ROM hashes match:
https://datomatic.no-intro.org/index.php?page=show_record&s=23&n=0112
https://datomatic.no-intro.org/index.php?page=show_record&s=23&n=1886

CobaltoSketcherino, 2023-11-03:
I did some digging, even double-checking that I had everything sorted out ROM hash wise, turns out my gba_db.bin was the culprit. I think I must've gotten an old version of that somehow. Sorry for the trouble, and I absolutely LOVE the work you're doing!

profi200, 2023-11-03:
Oof, i really should add a header with hash to the db format so it can be verified. I already saw issues like that coming.

closed 2023-11-03 by profi200