profi200/open_agb_firm #146: Logging and replaying button inputs

exelotl, 2023-11-03:
When developing a GBA game it can be useful to record button inputs while testing, so that in the event of a rare bug you can deterministically replay the whole gameplay session up to that moment (to reproduce it or to check if it's been fixed in the latest build)

This is doable in mGBA with the new scripting feature, but less doable (or at least has more caveats) when testing the game on real hardware.

Would a feature like this be something `open_agb_firm` could support?

profi200, 2023-11-03:
The problem i see coming is that i will run out of memory quickly. There is less than 512 KiB left not counting VRAM. If i record button inputs to the SD there is a little problem too. SD card writes will block all my threads the way it's implemented right now and that is pretty bad.

Another problem is i can only record inputs once per frame but that does not necessarily reflect what the game sees because the game samples button inputs directly from hardware (unless button remapping is enabled) and it can do that at any point within the frame time.

I will have to think about ways to do this that does not interfere with any other feature.

exelotl, 2023-11-04:
Thanks for the insight!

I had a shot at implementing this myself before I saw your response. Writing directly to SD card didn't seem to cause any problems but I guess I was just lucky (I only played for 5 minutes). But, my file was blank because `lgy11->padcnt` seemingly always gives zero, RIP. :(

With naïvely writing to RAM you'd be able to record for over an hour before 512KiB is used, but if you were to log e.g. keypad state + duration of that state in frames, you could probably play for hours before running out.

The input timing thing is troublesome yeah. I guess while the recording feature is active you'd have to enable remapping for every single button, which would require an extra layer over the existing remapping feature?

profi200, 2023-11-04:
lgy11->padcnt is not the correct reg. You are looking for hidKeysHeld() which spits out the buttons held and is updated once per frame (main.c).

As for SD write: It should cause issues if the write happens at the wrong time. What usually happens is the video output glitches and you get a similar effect as with AGB_FIRM where the whole picture is shifted and wraps around.