profi200/open_agb_firm #147: Override scaling mode using keypresses on startup

patters-match, 2023-11-04:
Sometimes I like to play a game hardware scaled, but sometimes not. Editing a text file on-device to change this behaviour is a bit tricky. Would it be possible for open_agb_firm to copy the behaviour of the Virtual Console emulators which will display unscaled if Start or Select are pressed during startup?

profi200, 2023-11-04:
It's not really mentioned in the README but per game configs can override all settings from the global, default config if that helps. Otherwise i think it makes more sense to handle this via a GUI eventually.

patters-match, 2023-11-05:
I do think the Start/Select pattern would be ideal to force unscaled, since that's the established pattern (even for Twilight Menu++) and it's probably a lot easier to implement than a full GUI will be. Whether or not to use a different key combo to force bilinear Vs hardware is kind of a secondary concern. I'd judge it's possibly less likely a person would want to alter that, as people tend to be fans of one mode over the other.