profi200/open_agb_firm #151: Is it possible to replace agb.firm using Luma?

Linx-ESP, 2023-12-24:
In the [Luma wiki](https://github.com/LumaTeam/Luma3DS/wiki/Optional-features) it has a part for "Enable loading external FIRMs and modules" and it includes agb.firm
Should renaming it to agb.firm and putting into luma/sysmodules just work? Would be possible in the future?

profi200, 2023-12-24:
While it will likely work it will just boot into the file browser just as if you booted the FIRM file directly. I have not tested this. There is currently no way to make it boot the VC titles directly as it doesn't have code to access NAND/eMMC.

It's possible to place autoboot.txt on the SD and make it boot into a game stored on the SD.

mirrorbender, 2024-01-07:
I tried renaming to agb.firm and putting it into luma/sysmodules and it didn't seem to do anything. GBA VC titles launched normally.

profi200, 2024-01-07:
You have to turn on loading external FIRMs/sysmodules. https://github.com/LumaTeam/Luma3DS/wiki/Optional-features#config-menu-options

mirrorbender, 2024-01-08:
I do have loading external FIRMs/sysmodules turned on in the Luma config menu. Nada. The expected behavior would be that open_agb_firm would boot just to the rom selector when you try to launch any GBA VC title, right?