profi200/open_agb_firm #152: Accidentally replaced FIRM0 with open_agb_firm

Sana2814, 2024-01-05:
How can I reverse this, I have tried to get ntrboot going but it just immediately boots to open_agb_firm instead of the B9S installer. Please help

profi200, 2024-01-05:
No, if you have a working ntrboot flashcard and it's flashed with ntrboot it should boot boot.firm from your 3DS SD card. Other than using ntrboot there isn't really a way to recover this. Maybe a hardmod.

irowiki, 2024-01-05:
> How can I reverse this, I have tried to get ntrboot going but it just immediately boots to open_agb_firm instead of the B9S installer. Please help

If it's not booting from ntrboot then either something is up with your flash cart, your magnet, your key combo or you're missing the proper boot.firm on the SD

Sana2814, 2024-01-05:
I got something else to happen but my hand slipped while doing it, going to keep trying but this is definitely going to be a pain in the ass, not least because my 3ds's card slot is really inconsistent.