profi200/open_agb_firm #153: Game & Watch Gallery 4/Advance doesn't save

JORGETECH, 2024-01-08:
It seems like the game "Game & Watch Gallery 4" (Game & Watch Gallery in Europe, I'm using this version) has issues with saving the game. One easy way to test this is to start a game and pause it before shutting down the 3DS, in a real GBA the progress of the game is saved up to the pause point so you can continue your progress, however this doesn't happen for me, and I loose all the progress I make in the games.

According to [MAME's GBA database](https://raw.githubusercontent.com/mamedev/mame/master/hash/gba.xml) the game should use SRAM chips, so I don't think it's related to the EEPROM limitation.

profi200, 2024-01-10:
Can't reproduce. The game saves as expected. Make sure [the ROM hash matches](https://datomatic.no-intro.org/index.php?page=show_record&s=23&n=0643) and use save type database from the oaf version you currently use. If in doubt just redownload and unpack the latest release overwriting all files.

JORGETECH, 2024-01-10:
I followed your advice and the problem was solved by updating open_abg_firm. I'm not sure when I updated previously, but my guess is that I didn't have the latest version of either the save game database or the main program. It wasn't the ROM dump I was using since the SHA-256 hash was the same as dat-o-matic's one.

JORGETECH, 2024-01-10:
Closing the issue, since it was my fault.

closed 2024-01-10 by JORGETECH