profi200/open_agb_firm #154: Super Mario Advance 2 and Super Mario Advance 3 Europe versions save type not being detected properly.

lbezos10, 2024-01-19:
Super Mario Advance 2 and Super Mario Advance 3 European releases when ran on open_agb_firm says: "Save is Corrupted" in game.

After some investigation, the save size is incorrect, 512 KiB, which equates to a save type of SRAM 256k, type 14, throwing up the error when in game.

After noticing that Super Mario Advance and Super Mario Advance 4 is 128KiB or in practice 64k EEPROM, i changed the save type on the aformentioned non-working games to be type 2 and, voilá, the game boots. :)

Long story short, i have discovered the temporary fix to those games, hopefully, the developers can add those games in the gba_db.bin.

Regards.

 - LBezos10.

profi200, 2024-01-19:
Update to the latest release and overwrite all files including the database. These games do work.

lbezos10, 2024-01-21:
I am sorry for not clarifying. I pull the release source code and compile it on a Debian VM and only get the binaries if the source code isn't stable/doesn't generate proper binaries from your repo. I am going to update it and compile it.

 Many thanks.

 - lbezos10

lbezos10, 2024-01-21:
I did a fresh pull of the source code, used `make release` tested on a fresh save and the issue appears. I am going to test the binaries for certainty. I would be happy to provide evidence in the case of pulling the source code.

Sorry for the incovenience.

 - LBezos10

lbezos10, 2024-01-21:
I used the available binaries. Still, no luck. However, i do suspect the reasoning behind the exception in the autodetection is that the ROMs were trimmed by GodMode9. Sorry for the confusion. It is up to you if you want to add the exceptions or not. I will not demand anything since it was my error. :)

Have a good day.

 - LBezos10

lbezos10, 2024-01-21:
Well, another error of mine... (Hopefully you already realized) I looked at the wrong size. They don't default to 512KiB, they default to 512 Bytes, which is 8k EEPROM. I did confirm that the games use 64k EEPROM (Type 2) instead of 8k. (Type 0) It is also good to mention that Type 1, doesn't create the save. Again, these are trimmed ROMs. I will test the untrimmed ones for more certainty.

lbezos10, 2024-01-21:
I confirmed that the reasoning behind it is that they were GM9 trimmed. Again, sorry, i will not demand anything from you.

Many apologizes.

 - LBezos10

lbezos10, 2024-01-21:
I am going to close the issue if i have the ability to. Since i forgot.

closed 2024-01-21 by lbezos10

profi200, 2024-01-21:
That is strange. I have code for reconstructing the original ROM (untrimming) in RAM. It *should* work. I will investigate.

profi200, 2024-01-21:
GodMode9 trims correctly. The issue is that these games use zeros for padding the ROM but oaf assumes 0xFFs. I'm not quite sure how to fix that. We will see.

lbezos10, 2024-01-21:
Can you tell me what is oaf in this context?

lbezos10, 2024-01-21:
You don't need to tell me now. It's just curiosity of mine.