profi200/open_agb_firm #155: How do I remap buttons?

tairisu, 2024-01-20:
Attempting to remap buttons seems to have zero effect in-game for me, here's what my config.ini looks like.

[general]
backlight=117
backlightSteps=5
directBoot=false
useGbaDb=true

[video]
scaler=2
gbaGamma=2.2
lcdGamma=2.2
contrast=1.0
brightness=0.0

[audio]
audioOut=0
volume=127

[advanced]
saveOverride=false
defaultSave=14

Y=B
B=A

profi200, 2024-01-20:
[input]
Y=B
B=A

tairisu, 2024-01-21:
> [input] Y=B B=A

thank you! i tried to add the [input] before but i probably added a space between the [input] and the buttons i wanted to remap. the issue is solved so i should close this but i would also like to ask if there's a way (or will be a way in the future) to enable ghosting?

profi200, 2024-01-21:
For what it's worth i just noticed the above mapping was the wrong way around. You map 3DS buttons to GBA buttons. See the README for details.

"ghosting" the same way as official GBA VC? No, not planned. But inter-frame blending (the same as what mGBA uses) is planned to remove flickering.

Cherriesd27-wq, 2026-02-13:
> > [input] Y=B B=A
> 
> thank you! i tried to add the [input] before but i probably added a space between the [input] and the buttons i wanted to remap. the issue is solved so i should close this but i would also like to ask if there's a way (or will be a way in the future) to enable ghosting?

i was wondering what you wrote that worked ? Ive been trying to change my mapping like you