profi200/open_agb_firm #157: Why are reboots are required for switching between games.

westleyaaj, 2024-01-27:
In the read me it says do to the gba hardware Reboots are required for switching between games. Why is that.

profi200, 2024-01-27:
Because the RAM the ROM is loaded to and other RAM becomes inaccessible permanently. Only a hardware reboot can bring them back.

westleyaaj, 2024-01-27:
Why is it inaccessible permanently.

Masamune3210, 2024-01-27:
It's just how the backward compatibility modes are implemented. Once the system is switched into either AGB_FIRM or TWL_FIRM, the only way back out is to fully reboot the system.

profi200, 2024-01-28:
As said, this is how the hardware works. I don't think there is any way around this. Once it's in GBA or DS(i) mode you can't go back. If there is a way to go back to 3DS mode then i/many others have not found it.