profi200/open_agb_firm #159: Support Vast Fame/Unlicensed cartridges?

N-SPC700, 2024-01-30:
Vast Fame, an unlicensed developer, made quite a few unlicensed GBA titles, all of which work in emulators, but fail to run on a 3DS. It's been a while since I looked into the specifics, but I believe the cart has some sort of special copy protection or something of that kind that emulators seem to implement, but open_agb_firm does not. I suppose it is not a huge loss if these cannot be supported, but since the games do run on a real console and in GBA emulators just fine, I think it would be great if I could play them on a 3DS as well.

profi200, 2024-01-30:
If this uses some kind of hardware copy protection then that's a no. oaf is not emulating anything. The game runs on hardware natively. I can only support what the hardware supports.

That being said if it's just copy protection you can try to patch the checks out.

N-SPC700, 2024-02-04:
I haven't tested different save types yet. I'm not sure which would work with these games. I'll see if that does something for it.

The compatibility does seem to be emulator-related, as [mGBA has dedicated explicit support.](https://github.com/mgba-emu/mgba/blob/bba2c131a172d6a209e9a96b827e670723deb0af/include/mgba/internal/gba/cart/vfame.h#L24) This tells me it may not be supported at all unless the copy protection is cracked, but if there's a workaround with save types, I will definitely give that a shot. I'll report results and see if I get anything working.