profi200/open_agb_firm #160: Firered Rom hacks

sneakyassasin02, 2024-02-02:
Im having an issue where Firered based rom hacks keep looping at the "Last Time User saved in Pallet Town"

Uploading VID_20240202_154542065.mp4…

Dartz150, 2024-02-03:
Many Pokémon rom hacks are known to have issues on real GBA hardware, and `open_agb_firm` runs on real GBA hardware that is inside your 3DS, so this also applies for it.

Try to run those rom hacks on mGBA, the most accurate GBA available emulator, and if you get the same results, then the issue is the rom hack itself. and it should be reported to the Romack maker instead.

profi200, 2024-02-03:
Not sure i understand the problem but sounds like RTC issues. Keep in mind i don't quite handle RTC right yet. It seems many ROM hacks need the RTC offset instead of making the RTC jump to the current time and date every boot.

If it's a savegame issue then try a different save type (backup your save first).