profi200/open_agb_firm #161: Just a request 😁❤️

DenizAydin01, 2024-02-13:
You guys are actually doing an amazing job for small 3ds modding community. Like even non-nintendo and weaker devices got better gba emulators such as psp, etc. Idrk why mgba even exists tbh its a project thats born death without old 3ds support at the first place. This mod is perfect and the best way to play gba games on 3ds. This project only needs rom swapping feature without rebooting the console everytime right now. Im using this on an old 3ds xl and works amazing and the new brightness control feature works also great with key combos. Thanks for the great work guys, bless yall 🙏🙏.

Masamune3210, 2024-02-14:
ROM swapping isnt doable at least currently due to how the different firmwares work with regards to the system. It would take somehow being able to at least mount the storage and read out a file, something that AGB_FIRM just can't do

DenizAydin01, 2024-02-14:
> ROM swapping isnt doable at least currently due to how the different firmwares work with regards to the system. It would take somehow being able to at least mount the storage and read out a file, someth

> ROM swapping isnt doable at least currently due to how the different firmwares work with regards to the system. It would take somehow being able to at least mount the storage and read out a file, something that AGB_FIRM just can't do

Thanks for the reply, i have two questions. Does games that requires gyroscope, works with using this method? Such as warioware twisted, etc. And the another question is, can you also play gameboy color games with this mod, since it works like a normal gba, thanks

Masamune3210, 2024-02-15:
GBC/GB would require the GBC cpu or emulation of it, which afaik, AGB_FIRM does not do. GB/GBC are handled as normal Virtual Console-esque emulators that run as normal titles when booted into NATIVE_FIRM i.e. when the 3DS is running normally

profi200, 2024-02-15:
As for booting back to the ROM selector i explained here why it's a problem:
https://github.com/profi200/open_agb_firm/issues/157#issuecomment-1913143485

There is no way around a hardware reboot but i'm still looking for ways to reboot directly into oaf again without user interaction.

DenizAydin01, 2024-02-15:
Thanks for the answers guys, and @profi200, i hope you will figure out how to force it to reboot into rom selector again. Much blessings ❤️🙏

closed 2024-02-21 by DenizAydin01

profi200, 2024-02-22:
That menu is a virtual console menu provided by the emulator. The classic NES games are all emulators. That it freezes is currently normal because sleep mode is not fully implemented. Press the buttons as instructed and it should run again.