profi200/open_agb_firm #164: Metal Slug Advance (can't save)

DenizAydin01, 2024-02-21:
So i was playing games like warioland 4 etc. They seemed to save ok, but when i try to save in metal slug advance it says save aborted. I patched the rom with sram patcher (gbata). But it gave the same error again so is there any solves?

profi200, 2024-02-22:
Do not use SRAM patches. It doesn't work and doesn't solve most issues. Make sure the hashes of the ROMs that do not work [match known good dumps](https://datomatic.no-intro.org/index.php?page=search&s=23). Also make sure you use the latest release and you overwrote the save type database on your SD card.

profi200, 2024-02-24:
I would prefer if you don't link to ROMs or mention ROM sites.

I don't know why Doom still doesn't save. As said make sure to use the latest release and overwrite the save type database too (when extracting merge files and replace). I will test this game later but i'm pretty sure it works.

A R4 card will not solve the issues because then you are using GBARunner2 and it has lots of compatibility issues. It's not emulation but it's also not native hardware.

DenizAydin01, 2024-02-24:
> I would prefer if you don't link to ROMs or mention ROM sites.
>
> I don't know why Doom still doesn't save. As said make sure to use the latest release and overwrite the save type database too (when [...]

There is this one called gbaemu4ds, they say it works a lot better than gbarunner2. And plays the games in .mb format

profi200, 2024-02-24:
It has not been updated in like a decade and it was very proof of concept. GBARunner2 surpassed it in any way. And the work in progress GBARunner3 is again quite a bit better than GBARunner2. Nothing beats native hardware however.

profi200, 2024-02-24:
I checked regarding Doom (0154 - Doom (USA, Europe)) and it saves fine. I had to replace the save type database on my end however.

DenizAydin01, 2024-02-25:
> I checked regarding Doom (0154 - Doom (USA, Europe)) and it saves fine. I had to replace the save type database on my end however.

Ion know how to replace the save type databese 😭😭 im pretty noob sry