profi200/open_agb_firm #166: Shader?

ErikaBiohazard, 2024-02-28:
How do I make it look the same in this image?
Open agb firm

[image]

DenizAydin01, 2024-02-28:
Whats the point of getting the gameplay even smaller 💀💀

profi200, 2024-02-29:
I'm not sure what you are asking for. Post processing shaders? Borders? Borders are already supported. See the README and issue #44.

ErikaBiohazard, 2024-02-29:
> I'm not sure what you are asking for. Post processing shaders? Borders? Borders are already supported. See the README and issue #44.

I would like to know if you can leave the image with scanlines or some shader

Dartz150, 2024-02-29:
> > I'm not sure what you are asking for. Post processing shaders? Borders? Borders are already supported. See the README and issue #44.
> 
> I would like to know if you can leave the image with scanlines or some shader

Not possible or practical. Besides this requiring fragment shaders, which are not supported by the 3DS, the screen resolution isn't high enough to provide a good "scanline/pixel grid" effect. That would require at least 4x the GBA resolution, and the 3DS is only 1.5x the GBA resolution. Also, `open_agb_firm` 1:1 resolution mode + borders is already in pixel perfect resolution, so it shouldn't need a shader on top.

Only color correction is being worked on in closed betas, but it uses a completely different approach and is aimed to make games look as how they would in a real GBA console, fixing horrible oversaturated colors in early titles such as Golden Sun or Donkey Kong.

ErikaBiohazard, 2024-02-29:
> > > Não tenho certeza do que você está pedindo. Pós-processamento de shaders? Fronteiras? As fronteiras já são suportadas. Consulte o README e a edição [nº 44](https://github.com/profi200/open_agb_firm/issues/44) .
> > 
> > 
> > Gostaria de saber se é possível deixar a imagem com scanlines ou algum shader
> 
> Não é possível ou prático. Além de exigir fragment shaders, que não são suportados pelo 3DS, a resolução da tela não é alta o suficiente para fornecer um bom efeito de "scanline/pixel grid". Isso exigiria pelo menos 4x a resolução do GBA, e o 3DS tem apenas 1,5x a resolução do GBA. Além disso, `open_agb_firm`o modo de resolução 1:1 + bordas já está em resolução perfeita de pixels, portanto não deve precisar de um shader na parte superior.
> 
> Apenas a correção de cores está sendo trabalhada em betas fechados, mas usa uma abordagem completamente diferente e tem como objetivo fazer com que os jogos tenham a aparência de um console GBA real, corrigindo cores horríveis e supersaturadas em títulos anteriores, como Golden Sun ou Donkey Kong.

Thank you very much for the explanation. This is the best way to play GBA on 3DS and I really loved it ❤️

closed 2024-02-29 by ErikaBiohazard