profi200/open_agb_firm #171: Capture Card Installation Glitchy Video

papajefe, 2024-03-26:
I have a New 3DS XL that is hardmodded with a capture card from loopy (3dscapture.com)

When using open_agb_firm, I get a black feed with the occasional flicker of the game image.

Luma\GodMode9 and all stock 3DS screen contents transmits to my PC successfully, but open_agb_firm specifically appears to have issues. I've tried tweaking gamma settings, and both 1:1 and hardware scaling.

I reported on Loopy's support forum as well, because I'm guessing this is more of a driver\software-side fix on his end.

However, I reported here in case you might have a configuration setting that would make a difference?

Example image of "flicker" below:

[image]

profi200, 2024-03-27:
I have seen issues like this before.

2 things of note:
* The top LCD does not run at a fixed refresh rate but it's alternating between 2 ever so slightly slower and faster settings than GBA output. This is needed to archive perfect output without skipping or duplicating frames. TWL_FIRM and AGB_FIRM (official DS and GBA mode) do the same.
* The bottom LCD does not have the same correction so both LCDs run out of sync.

I suspect it's the later the capture card doesn't like. Maybe it synchronizes with the bottom LCD or something.

profi200, 2024-03-27:
Does this build work better? It will keep both LCDs synchronized. If it does it's a problem with the capture card. It should not assume that both LCDs are always synchronized.

[open_agb_firm_lcd_sync.firm.zip](https://github.com/profi200/open_agb_firm/files/14773474/open_agb_firm_lcd_sync.firm.zip)

papajefe, 2024-03-27:
This did indeed completely fix the issue. That was so fast, thank you so much!

I'm guessing the downside here, is that the bottom screen is now powered-on and using battery when it doesn't need to be.

The good news is, I can use this separate build for recording and the normal release when playing on my own. Great stuff.

profi200, 2024-03-28:
Use the latest build from the Actions tab. The above one is just a test build.

The bottom LCD was always on because the MCU can only turn on or off both at a time. The difference is they are now always running equally fast. As said above this is still the fault of the capture card for assuming that the LCDs will always run at the same speed when this is not guaranteed. The hardware has no issue with this. It's just the capture card that goes nuts.