profi200/open_agb_firm #180: Screen brightness thing

DenizAydin01, 2024-04-27:
When on old 3ds device, Opened rosalina menu. In change screen brightness sub menu, you can see screen luminance values between 16 and 142. So i wonder why does it say minimum 20 in readme.md for old 3ds devices. Because 16 and 142 isnt just there like sembolic thing it actually works and i can see the screen getting slowly dimmer between 16 and 20.

profi200, 2024-04-28:
I'm not sure where the old limits come from but the old brightness system is based on PWM duty cycles which is not great but was easy to implement. The new system used in the libn3ds_gfx_lgycap branch uses luminance and requires hardware calibration. Current Luma versions also show luminance if i remember correctly.

DenizAydin01, 2024-04-28:
> I'm not sure where the old limits come from but the old brightness system is based on PWM duty cycles which is not great but was easy to implement. The new system used in the libn3ds_gfx_lgycap [...]

So is it okay if i leave the brightness config set to 16 in config.ini file for an old 3ds

profi200, 2024-04-28:
It will likely get set back to 20 somewhere. I don't even know if we are talking about luminance or PWM duty cycle here. It does matter because brightness is not linear.