profi200/open_agb_firm #182: Save Data

DenizAydin01, 2024-05-04:
So i downloaded a save file with everythng unlocked, for mortal kombat: tournament edition, the file was in .sav format as it supposed to be and i named the save file same as the game file. Everythings good. But when booting the game, it just wont read the save data, however when using the same save file with an emulator on android, it just plays fine.

Dartz150, 2024-05-04:
That game uses save type EEPROM_4K, which needs to be converted to its proper format with non-reversed endianness. You need to use the EEPROM fixer as stated in the readme.

DenizAydin01, 2024-05-04:
> That game uses save type EEPROM_4K, which needs to be converted to its proper format with non-reversed endianness. You need to use the EEPROM fixer as stated in the readme.

Oh it worked, thanks a lot G, now i can enjoy my  all unlocked pixelated 120p characters.😂

closed 2024-05-04 by profi200