profi200/open_agb_firm #183: Save files incorrect time and date

Adnan12349, 2024-05-17:
When creating a game save, it saves it with an incorrect time and date. For example, I saved at 5/17/2024 21:00, but it shows as 5/22/2022 00:00 instead, which is totally random and wrong. It should have adapted to the console's clock time and date.

profi200, 2024-05-18:
Your consoles RTC is not set correctly. The OS running on 3DS normally uses an offset from the time and date set in the RTC but it's way too much effort to get this offset from the system just for time and date.

You can open Rosalina while in HOME menu and use the options to set the time/date offset to 0 and then use the NTP feature to correct the time and date.

Adnan12349, 2024-05-19:
This did not solve the issue. I set the RTC on my console correctly and set the time/date offset to 0, then used the NTP feature to correct the time and date on the Rosalina menu. However, when creating a save, it still shows as 5/22/2022 00:00. I tried changing my console time to a random date, but it is stuck at 5/22/2022 00:00. Is it saving the time correctly for you?

Dartz150, 2024-05-19:
> This did not solve the issue. I set the RTC on my console correctly and set the time/date offset to 0, then used the NTP feature to correct the time and date on the Rosalina menu. However, when [...]

What game are you trying to run that doesn't work?

Adnan12349, 2024-05-19:
I tried Pokémon Emerald and Sonic Advance 3. When creating a save file, it always saves as 5/22/2022 00:00.

profi200, 2024-05-19:
Ah, are we talking about the in game clock or file creation dates? I don't have correct file creation dates implemented right now so it uses whatever i hardcode (usually the last time i updated the FatFs driver).

Adnan12349, 2024-05-19:
Got it, thanks for the clarification. I was actually referring to the file creation dates rather than the in-game clock. I usually keep my save files in sync, so it's important for them to have the correct time and date.

profi200, 2024-05-20:
I may add this later but first i need to come up with a way to provide the time and date to the FatFs driver on ARM9 because reading the time and date from the MCU is only possible on ARM11 the way my library works. But anyway, we will see.

profi200, 2024-05-20:
Well, i got it partially working. I had the idea to use the same RTC that GBA games use and forgot that some games reset the time and date. The access time seems to be a FatFs bug. It should at least set the access time to the creation time.

In order creation time, modified time and access time. "Heute" means today.
[image]