profi200/open_agb_firm #185: Extended Brightness Configuration Request

radicalten, 2024-05-25:
Hello,
Based on your reply to issue #180 Screen brightness thing : "The new system used in the libn3ds_gfx_lgycap branch uses luminance and requires hardware calibration. Current Luma versions also show luminance if i remember correctly.", is it possible to allow for extended backlight control (increasing the range from the default 16-142 luminance to an extended 1-172 luminance)?

I attempted to do this myself by forking the open_agb_firm repository and making some changes to the libn3ds_gfx_lgycap branch, but I haven't had any success increasing the luminance range yet. 

Any help with how to achieve extended luminance would be appreciated. 

p.s. Thank you for creating and maintaining this project!

profi200, 2024-05-26:
You will have to dump the most up to date of the 2 HWCAL files to /3ds/HWCAL.dat first so the brightness curve matches. Then there is a limit somewhere i have not removed yet if i recall correctly.

closed 2024-05-28 by radicalten

UltimateBatDave, 2024-12-26:
I'm sorry if I am just not understanding correctly, but I think both of my HWCAL.dat files have the same timestamp, in godmode9 at least, dated year 2001. Unfortunately it changes when I dump them, so am to assume it doesn't matter which one to rename HWCAL.dat and put into /3ds/ or does it need to be HWCAL1.dat, for example?

profi200, 2024-12-26:
It's this i'm referring to. https://www.3dbrew.org/wiki/Hardware_calibration#Header

`0xD 	0x1 	CAL revision (incremented each time the CAL file is updated)`

The file that has the biggest number is the most up to date one. I should make a GodMode9 script or something to dump it.

UltimateBatDave, 2024-12-26:
Thank you! If I got it right, my HWCAL0.dat has 0B and 1.dat 0A, so I renamed the 0.dat to HWCAL.dat and put into /3ds/.