profi200/open_agb_firm #187: Metroid: Zero Mission (Europe) (Virtual Console) Is Unplayable

emariz, 2024-05-26:
Most likely not an issue of Open AGB Firm, just a note to other users: **Metroid: Zero Mission (Europe) (Virtual Console)** seems to be unplayable; it halts at a white screen upon load.
The ROM's integrity was verified before testing the different save types, none worked.

Its page at No Intro's database:
https://datomatic.no-intro.org/index.php?page=show_record&s=23&n=x159

Granted, this is a **Trusted - Not Verified** dump, but other ROMs that halt at a white screen do play after the correct save type is selected.

profi200, 2024-05-31:
I keep getting distracted. I will take a look at this and the other issue later. In the meantime make sure the save type database is up to date.

profi200, 2024-06-01:
This VC dump is patched to high hell with emulator specific looking code + some extra headers/footers.

The first instruction in the ROM has been patched to jump to 0x8784000. At that offset i see instructions which are not valid on ARM7. For example:
`8784008:	ee801000 	cdp	0, 8, cr1, cr0, cr0, {0}`

This will likely never run correctly on hardware.

profi200, 2024-06-01:
With this patch the game seems to run fine. It removes the weird jump to 0x... so the invalid code never gets executed but i can't guarantee that they didn't add more weird patches.

Patch the game with armips. https://github.com/Kingcom/armips
```
.gba
.arm

; Helpful constants.
rom_start          equ 0x08000000
rom_entry_branch   equ rom_start
multiboot_branch   equ rom_start + 0xC0
input_file_name    equ "metroid_vc.gba"
output_file_name   equ "metroid_vc_patched.gba"

.create output_file_name, rom_start
.org rom_entry_branch
	b multiboot_branch

.incbin input_file_name, 4

.Close
```