profi200/open_agb_firm #189: [Request] 800px mode

mlgmaster666, 2024-05-31:
I was wondering if using the full 800px hz res was even possible on a .firm.

If possible, i think the scaling could look a lot better by having the top screen in 800x240 mode instead of the normal 400x240, i know this is possible on installed software, since the mario64 port did it, but i really dunno how hard it would be (if even possible) to implement on a .firm .

profi200, 2024-05-31:
The mode as in output is already implemented. The difficulty is that i can only scale this with the GPU since the hardware scaler is limited to 512x512 and i don't have citro3d available to do arbitrary scaling meaning it's a mess to even change anything in the GPU render pipeline.

It's on my list however when i tried out this scaling mode in mGBA it was not as impressive as i thought so it's not very high priority.

mlgmaster666, 2024-06-02:
I tried installing mgba, while i personally find it pretty noticeable in some games, like mario3, since it makes it a perfect integer 3x hz, i can understand why it wouldn't be a priority rn, since that does sound like a lot of work for something that is only a slight visual upgrade.

buymyhubs, 2025-03-19:
Seconded, this would be a really nice upgrade if possible. I personally find mGBA's 800px gba scaling to be miles better in practice