profi200/open_agb_firm #190: Failed Saving at Random (Pokemon Emerald)

SSeptic, 2024-06-25:
I have played Pokémon Emerald for probably around a total of 21 hours thus far. However, I have run into an issue on three occasions wherein the game does not save correctly, and despite saving, enduring the save animation and then waiting a few seconds to be sure, loading back in shows that the save was not overwritten and I am one save behind. I make this issue because I believe it to be different from the aforementioned issue in the readme, as saving works most every time. I do not know what causes this bug and attempting to replicate it brings only successful saves.

profi200, 2024-06-25:
I have yet to hear anyone having issues with savegames and i have personally played games for many hours as well without issues. It's likely that either your SD card is corrupted in some way or it's failing. In any case, immediately make a backup of your data. You can try to reformat your SD. https://3ds.hacks.guide/formatting-sd-(windows)

SSeptic, 2024-06-25:
Well shit, thanks. I’ll try that and if anything crops up again I’ll reopen the issue maybe

closed 2024-06-25 by SSeptic