profi200/open_agb_firm #195: Stuck at GBA Boot Screen with no nintendo logo

SebRock124, 2024-07-31:
Title Says it all...

Dartz150, 2024-07-31:
What are you trying to run? every GBA game on the library and most of the rom hacks should work fine (Except 64MB GBA video roms)

SebRock124, 2024-07-31:
Ok nevermind... just found out it was because I was choosing the files with a broken name. There is a duplicate of all my files. It works now though. Just loaded up emerald fine.

closed 2024-07-31 by SebRock124

profi200, 2024-07-31:
Which duplicates? Is this yet another case where Mac OS creates meta data files that are hard to filter out?

SebRock124, 2024-07-31:
Yes...

profi200, 2024-07-31:
Try this nightly build. It should filter out these files.
https://github.com/profi200/open_agb_firm/actions/runs/10182033472

SebRock124, 2024-07-31:
I just deleted the files in windows

SebRock124, 2024-07-31:
But thank you for helping. Btw awesome project. I use the nightly build from when I submitted this and it works great!