profi200/open_agb_firm #197: Game menu sometimes chooses the wrong game

manu-romero-411, 2024-08-04:
On the latest beta version, if starting scrolling on the game menu from bottom upwards, it can happen that the game isn't correctly chosen. Instead, `open_agb_firm` will load the game which was above on the list.

This isn't happening in every case. If my game list is considerably big, and I start scrolling from top downwards, the behaviour is the expected: the chosen ROM will be the loaded ROM. I've noticed that when the menu scrolls from bottom to top the bug isn't always occurring.

On the previous alpha releases this bug was inexistent. All games loaded fine.

profi200, 2024-08-05:
I can't reproduce this behavior. You need to be more specific on the steps to reproduce this.

manu-romero-411, 2024-08-23:
Sorry, I wasn't too specific.

My system is a New 2DS XL, if it's useful for you to know. As an example, we consider three games in the menu:

```
  GameA.gba
> GameB.gba
  GameC.gba
```
We have `GameB.gba` highlighted on the menu, as noted by the `>`.
What's expected: when pressing the A button, `GameB.gba` loads.
What (often) happens: when pressing the A button, `GameA.gba` loads instead of the highlighted `GameB.gba`.

As I wrote on my previous message, this issue isn't occurring always. My game list is rather vast, around 120 ROMs, and is easier to reproduce the bug for me if I scroll to bottom. On a less populated game list, where scroll isn't needed to see all ROMs, the issue is harder to reproduce. 

As for now, I'm running older version which unfortunately lacks your latest improvements.

profi200, 2024-08-23:
No matter what i try i can't reproduce this behavior with the latest release even in a folder that has more entries than fit on the screen.

JoLiKMC, 2024-08-25:
@manu-romero-411
I have an incredibly *basic* question, but hear me out.  You didn't happen to rebind any of the controls in the main `config.ini` file, did you?  Specifically, the `A` button?  I'm not sure is rebound controls affects the load menu, but that's the only thing I can think of that could be happening.

profi200, 2024-08-25:
Button remapping is done in hardware and only affects the GBA hardware.

GandalfTheFray, 2025-02-20:
This is happening to me to, but its interesting because its not every game, I believe there is a single bad rom file throwing off the index or something somehow. Games at the top of the list load fine, but games towards the bottom seem to load the incorrect game.

Korkd, 2025-10-23:
Spent a bunch of time trying to understand this problem and have a theory.

In testing, this only occurs when there are 26 or more files in the directory. It appears to be a display issue and my best guess is this particular line in https://github.com/profi200/open_agb_firm/blob/beta_2024-12-24/source/arm11/filebrowser.c:

`ee_printf("\x1b[%lu;H ", oldCursorPos - windowPos + 1);`

When pressing up and there are greater than 26 files, `windowPos` has a value of 2 or more. That means on the next iteration of the loop you wind up with an int overflow. This is further reinforced as with 25 files, pressing up functions correctly and results in a `windowPos` of 1 not causing the int overflow.