profi200/open_agb_firm #199: (Request) Game forwarders for 3DS home screen?

midtoe, 2024-08-20:
is it possible to somehow have forwarders that would reboot the console into open_agb_firm and pre select a game? it probably is but is outside the scope of the project, so I might need to badger someone who can code or just learn to code myself to make a separate program to do this.

profi200, 2024-08-21:
I forgot where but someone has already done something like this.

There is currently no easy way to implement forwarders (and i honestly don't understand their appeal). This feature needs support from the bootloader and b9s doesn't support this.

redunka-zver, 2024-08-22:
Yeah, there's a [proof-of-concept forwarder](https://github.com/ismailhasannnnnn/OPEN_AGB_FIRM_Forwarder), but one would need to be able to build stuff on their own in addition to having a suitable bootloader (like GM9 or fastboot3DS) to use it, as already said above.