profi200/open_agb_firm #202: [Suggestion] Rebindable Screen and Screenshot Combos

JoLiKMC, 2024-08-26:
## Suggestion

### What is the feature you're requesting?
I'd like to see the ability to rebind or remap the Screenshot and Screen Brightness combos.

### How would you like to see this feature implemented?
Pretty much the same way the current controls are implemented; through an editable field in the `config.ini` file.  Maybe something like:
`[SYSTEM]`
`SCREENSHOT=Y,SELECT`
`BRIGHTNESSUP=X,UP`
`BRIGHTNESSDOWN=X,DOWN`
`LCDOFF=X,LEFT`
`LCDON=X,RIGHT`
`SKIPPATCHES=X`

### Anything else you'd like to say?
I think this would be handy for people who, well, want to rebind their "system" button controls.  For example, someone might want to bind the Screenshot combo to `L` and `SELECT`, or set LCD Off to `R`, `Y`, and `B`, or what have you.  Just a thought.

### Duplicate Issues
- #172

profi200, 2024-08-26:
The long term goal was to move brightness into a menu. For screenshots i probably have to keep a button combination since it's quite hard to trigger a screenshot from a menu at the right moment. I will take a look but it's low priority since there are other important issues like RTC offset support.

JoLiKMC, 2024-08-27:
Nah, that's understandable.  And this obviously isn't stopping me from enjoying your work!  It's just a (far) future consideration.  👍

Thank you for your time!