profi200/open_agb_firm #208: Super Mario Advance 4 crashes

StariaRose, 2024-10-14:
Apologies if this was already reported or is a known issue, I didn't find anything about it online.

Specs: Nintendo 3DS XL (old, not new). Latest CFW and open_agb_firm release.

Steps to reproduce:
1. Download SMA4 and put it on the SD card.
2. Load firmware, run game.
3. Play for a good bit.

Expected Result: Game should work normally. 

Actual result: Game screen goes entirely black after a bit. I've noticed it within 1-2 minutes. The game seems to be working in the background, the audio continues, I tried pausing and moving around and heard appropriate sounds etc..

Other Notes: This happened with SMA4's original release and VC/Switch Online releases with the E-Reader levels in them.

profi200, 2024-10-14:
Verify the hash against [this database](https://datomatic.no-intro.org/index.php?page=search&s=23) and make sure the save file is not corrupted. If it matches you may be one of the unlucky people to have defective hardware. @Dartz150 had 2 of them and was able to fix them by reflowing his FCRAM chips.

StariaRose, 2024-10-16:
Ah yeah, it matched all right. Guess I'm one of the unlucky ones. No clue how to reflow an FCRAM chip lol. Thanks anyways!

profi200, 2024-10-17:
It's not clear to us how this problem even happens because affected consoles show no signs of RAM failure in 3DS mode. Maybe it just can't keep up with the timings however then a reflow should not make a difference. ¯\_(ツ)_/¯

StariaRose, 2024-10-18:
Yeah... most GBA games run just fine, so this felt weird to me.

profi200, 2024-10-18:
Is it reproducable and is there a specific place in the game where this happens often? I could try this myself on a known good 3DS later.

vpelletier, 2024-10-18:
Lurker here, with another idea: there seem to exist several [test cartridges](https://tcrf.net/AGS_Aging_Cartridge) either intended to test actual hardware (original GBAs, not DS/3DS backward compatibility) or to test emulator compatibility. Maybe one can be used as a memtest ?

...or maybe a memtest can be somehow implemented ?

StariaRose, 2024-10-18:
> Is it reproducable and is there a specific place in the game where this happens often? I could try this myself on a known good 3DS later.

It's reproducible across the original release, the Virtual Console release and the Switch release. It happens not in a specific place, but at a specific time, about a few minutes in. At one point, this was in the E-Reader's first level. Another case, it was in the E-Reader castle thingie where I was idling for a bit. Another, in level 1-1 while I was going slow. Finally, in level 1-4 when I was going fast. I noticed _around_ the same time ish for each, though I didn't measure.

> Lurker here, with another idea: there seem to exist several [test cartridges](https://tcrf.net/AGS_Aging_Cartridge) either intended to test actual hardware (original GBAs, not DS/3DS backward [...]

Actually, that's interesting. I have no idea why/how this is a memory issue at all XD (I'm not super knowledgeable about low level stuffs). No 3DS game has had this issue, I'm trying to remember if I had the issue with Sonic Rush for DS or if I'm misremembering. I'll try it and lyk!

profi200, 2024-10-30:
Ah, i forgot about this issue because i had too many other things in my head the last few weeks. I will test that game probably this evening. If it does crash however i can't promise a fix because this is extremely hard to debug. But at least there are only few things that could make it crash.

profi200, 2024-10-31:
I played through the entire first world with a fresh savegame and no hangs, glitches or blackscreens. So yeah, i can't reproduce this.

cotodevel, 2024-11-02:
wow, this caught my attention. I had no idea about 3DS VC behaving like that on certain units. 

If SMA4 behaves correctly on DS mode through GBARunner2/3 on said O3DS XL, then it's probably got to do with GBA Slot access timing issue interfering between both the DMA engines as well the FCRAM (as GBA's EWRAM). It's almost as if a DMA (3) device takes ownership and interlocks with FCRAM.

Maybe Nintendo knew about this and did patches ([by adjusting the FCRAM timings on GBA](url='https://problemkaputt.de/gbatek.htm#3dsmiscregisters')) on the fly to overcome the GBA timing issues across different FCRAM vendors.

This is pure conjeture, but maybe it'll fix things for some 3DS models.

profi200, 2024-11-03:
FCRAM is only used for the hardware cart emulator. That means every time the GBA hardware wants to read ROM it will actually read from FCRAM. That RAM is a little wonky in general with people starting to report actual defects like bit flips and the much more common one being broken lines (cracked solder balls ect.).

And there is this mystery of GBA mode breaking in random games at random places. For example i mentioned how Dartz150 had 2 broken units. They both passed RAM testing in 3DS mode but they both had weird bugs in GBA mode. Some Pacman game he had to test for GBARunner3 development had the sound setting stuck at off every time the game is booted. The other one was a background animation being frozen in a Mega Man game. Other than that the games behaved normally. All of these weird bugs did go away after he reflowed the FCRAM chips.

cotodevel, 2024-11-03:
And? FCRAM has different timings for a reason, to provide 3DS/DS/GBA backwards compatibility across different brands.

So my point stands. Adjusting FCRAM timings will fix issues in certain units where the FCRAM hardware may be wearing out or where there may be game [may freeze due to](https://web.archive.org/web/20150221062059/https://www.nintendo.com/consumer/systems/gameboy/trouble_specificgame.jsp#megaman) [GBA Slot2 timings](https://github.com/RockmanEXEZone/BN4-WoodMan-Fix/blob/master/source/src.asm).

cotodevel, 2024-11-03:
And? FCRAM has different timings for a reason, to provide 3DS/DS/GBA backwards compatibility across different brands.

So my point stands. Adjusting FCRAM timings will fix issues in certain units where the FCRAM hardware may be wearing out or where a game [may freeze due to](https://web.archive.org/web/20150221062059/https://www.nintendo.com/consumer/systems/gameboy/trouble_specificgame.jsp#megaman) [GBA Slot2 timings](https://github.com/RockmanEXEZone/BN4-WoodMan-Fix/blob/master/source/src.asm).

StariaRose, 2024-11-03:
If it's ok to ask, I wasn't able to find much about reflowing the FCRAM online. Would one of you know how to do it?

Either way, open_agb_firm is incredible work thus far. Thanks so much!

profi200, 2024-11-04:
I'm not going down the rabbit hole of patching games Supercard style when this will cause issues with games and it's not clear if this will fix the issue. Nintendo intended it to work without any modification to games.

---
> If it's ok to ask, I wasn't able to find much about reflowing the FCRAM online. Would one of you know how to do it?

The short version is you heat up the chip and board until the solder balls melt which is usually used to solve defects like cracked solder balls. If you want a full guide maybe @Dartz150 can help. Keep in mind it's not a beginners job. It's easy to damage your 3DS.