profi200/open_agb_firm #216: Latest update breaks per-game settings

TheGershon, 2024-12-24:
Release 2024-12-24 breaks existing per-game .ini configs. Both color profiles and savetypes seem not to work, even after updating the strings to the new format. Perhaps it's something to do with the new useSavesFolder setting?

profi200, 2024-12-25:
The configs need to be in the same folder as the save. This is behavior is hardcoded.

TheGershon, 2024-12-25:
They are, I haven't changed their location since before the update.

profi200, 2024-12-25:
Pretty sure i tested this and the per-game settings are still working for me. Obviously the ini files need to be updated to the new config format.

TheGershon, 2024-12-26:
~~I have saves and per-game configs in `microsd:\3ds\open_agb_firm\saves`, for example `microsd:\3ds\open_agb_firm\saves\RomName.sav` and `microsd:\3ds\open_agb_firm\saves\RomName.txt`. The config looks like this:~~

I just noticed that all of my per-game config files were turned into txt files! No idea how or when that happened, very embarrassing. Closing!

closed 2024-12-26 by TheGershon

Masamune3210, 2024-12-27:
The same reason all the other features that make no sense or are objectively hated are enabled by default, Microsoft thinking they know what people better than the people themselves do.