profi200/open_agb_firm #229: Per-game settings don't appear to work - user error?

genericmelancholy, 2025-03-20:
I've recently updated open_agb_firm to the latest version, which includes the config file changes. I'm trying to play SMA3 which uses the EEPROM_64K save type; this isn't detected by default so I need to manually set this type using a per-game config file, which I have placed in the saves folder. Alternatively I could use `saveOverride=true` (this works) although that wouldn't be ideal. The config file I've created has the same name as the ROM and looks as follows:

```
saveSlot=0
saveType=eeprom_64k
```

However, open_agb_firm appears not to detect the setting and so the game detects my save file as corrupted upon booting. Have I done something wrong in the config file?

Khramey, 2025-04-22:
I too can't make per-game setting to work too.

ChrisHighwind, 2025-04-29:
While I'm not playing SMA3, I am playing Shantae Advance Risky Revolution as the game has been getting into people's hands, and thus a dump is now floating around the Internet (won't say where, though), and the ROM points to the completely wrong save format, throwing off the autodetect and requiring me to have to manually override the save format. I was able to do so by using a per-game config file with the same name as my ROM and looking as follows:

`[game]
saveType=sram_256k`

It seems to work for me. Perhaps try adding "[game]" to the line before the settings? If that doesn't work, are you sure you've put the per-game config file in the game saves folder where it needs to go?

Khramey, 2025-04-30:
> [game]
> saveType=sram_256k

I add [game] as you suggested and it's work now, thanks.

Dartz150, 2025-04-30:
> While I'm not playing SMA3, I am playing Shantae Advance Risky Revolution as the game has been getting into people's hands, and thus a dump is now floating around the Internet (won't say where, [...]

This game doesn't have a standard GBA header, causing detection routines in some GBA emulators to fail. I made a patch to fix the header and the save string to make it work in GBARunner3, since we need an existing TID for the JIT in GBARunner.

 You can get the fix patch in the mentioned issue https://github.com/Gericom/GBARunner3/issues/185

Lombardi128, 2026-02-28:
I'm currently trying to get per game settings to work, and I've exhausted all possible user error that I can think of. I'm trying to play a patched version of Boktai 1 with a solar sensor hack. I know for a fact that the save type is EEPROM 8k because that's the version detected by GBA injectors and saving does work when injected as a VC title.

My game file is named "Boktai The Sun is in Your Hand (Sensor Hack)" with a ".gba" file extension.
My game ini file is named "Boktai The Sun is in Your Hand (Sensor Hack)" with a ".ini" file extension.

The only two lines in the ini file are these two:

[game]
saveType=eeprom_8k

The save always fails after setting the clock time and always fails after trying to start a new game file. These both succeed when using a VC inject of the game with EEPROM 8k save type. The ini file is located in the saves folder located at SD:/3ds/open_agb_firm/.

Another thing I tried was to set the saveOverride to True in the main config file located at SD:/3ds/open_agb_firm/. I launched the game in open_agb_firm, expecting to see a menu to select which save type to use for the selected file for the current session, but no menu appeared. I tried loading an unpatched version of the game and the save override menu did appear. It seems it won't appear for the patched version.

Any insight on what the problem may be would be helpful. Another thing I would like to ask is about RTC on EEPROM. I noticed that there were no save file options listed in the per game settings instructions for EEPROM saves that had an RTC option. The Boktai series uses EEPROM for save type, and also had a battery powered RTC to maintain the time of day which would be represented through graphical changes in game. Does open_agb_firm not support RTC on EEPROM?