profi200/open_agb_firm #233: Button remaps stopped working

PirateCoveMan, 2025-03-29:
I realized that the latest update changed some of the configuration keywords, so while I was able to get my borders to work again, I can't for the life of me figure out how to get the buttons to map the way I want. I follow the instructions exactly, but nothing changes. Did the input keywords change and you forgot to update the instructions?

LovGo, 2025-04-09:
Can you post your config file so that we can review it ? It is working fine on my side

PirateCoveMan, 2025-04-09:
Here you go. Are you sure that your end is the latest version too?
[image]

PirateCoveMan, 2025-04-09:
Yes, I was able to get the resolution and borders to scale down after I figured out the new keywords. The buttons are all that don't work

PirateCoveMan, 2025-04-10:
So, um, I just deleted the config files and started over from scratch, and suddenly it's working perfectly. I really don't know what the deal was, but I ain't complaining

Nezuhko, 2025-04-19:
hi, if I may ask what application is the one shown in the photo above?

PirateCoveMan, 2025-04-19:
Markor. Apparently it's not on Google Play and must be downloaded from GitHub