profi200/open_agb_firm #241: "Your saved data is corrupted" warning when pre-patching or trimming SMA2 or SMA3

Soto-Jnthan, 2025-06-04:
There are various ways in which loading Super Mario Advance 2 - Super Mario World (USA, Australia) and Super Mario Advance 3 - Yoshi's Island (USA) result in a "Your saved data is corrupted" warning:
- When loading pre-patched ROMs instead of placing the .ips files and letting open_agb patch them
- When loading the ROMs after being properly trimmed with no patches applied (I used NDSTokyoTrim v3.11).

It is weird that the ROMs work with no issues when the ips patches are applied by open_agb, but they do not work when exactly the same patches are applied beforehand and the patched ROMs are loaded instead. I have reproduced this issue with the color restoration patches of both games, which do not touch the game codes of the ROMs. This, in combination with the fact that just trimming the unmodified ROMs is enough to trigger this error, leads me to believe that maybe there might be something worth looking into.

I have tried with SMA1 and SMA4 and they work well pre-patched and trimmed. There might be other games affected, so I'll update you in the case I find other ones.

Soto-Jnthan, 2025-06-05:
Additionally, the same pre-patched and trimmed games having this issue work flawlessly on VBA-M, mGBA and even GBR2-3, so it must be related to open_agb_firm.

leleobhz, 2025-07-12:
@Soto-Jnthan hello!

At least for SMA3, https://github.com/profi200/open_agb_firm/issues/186 applies.

TL;DR: 

* Create a file `/3ds/open_agb_firm/saves/Super Mario Advance 3 - Yoshi's Island (USA) (Wii U Virtual Console).ini` (Considering my save is named `Super Mario Advance 3 - Yoshi's Island (USA) (Wii U Virtual Console).sav` - It needs to be the same name of ROM (And usually .sav) but with .ini
* Paste following content:
```ini
[game]
saveType=eeprom_64k
```
* Save the file and remove .sav invalid save file
* Be happy

Masamune3210, 2025-07-13:
Likely, detection uses the hash of the file. Trimming it changes the file, which changes the hash. Hash no longer matches what is expected, so detection fails to select the right game in the database and the right save type as a result.

profi200, 2025-08-17:
The trimmed ROM should work because oaf restores the ROM padding in memory. Either there is a bug in oaf or NDSTokyoTrim. I will investigate.

Prepatched ROMs don't work because they don't match any hash in the database as mentioned above.

profi200, 2025-08-17:
The reason it's not working is that these games have zeros as padding but oaf assumes 0xFFs as padding which is true for almost all games. Since it reconstructs the ROM with the wrong padding bytes the hash won't match and the save type is not found in the database. There isn't really a good solution for this problem. Trying both types of paddings would just slowdown ROM loading.

Soto-Jnthan, 2025-08-17:
I appreciate that you have looked into it. Thanks! Would reading the game header of the ROM be considerably more difficult than performing the hash? I see that's the strategy taking by almost all emulators and HVs and that could solve both of the original problems at once